peterhinch/micropython-font-to-py #2: I'm working on a industrial scale using this display, I want to output realtime KG measurments to the display cause the contrast and clarity beat 7 segment displays + wifi from esp.

fourlexboehm, 2017-02-22:


fourlexboehm, 2017-02-22:
How the hell can I get real time larger fonts without saving every single KG value from 1 to 65000 as Image files?????

peterhinch, 2017-02-23:
I'm afraid you'll have to explain this in more detail. ``font_to_py.py`` is a generic utility not tied to any one display. It merely converts a font to Python sourcecode.

What display are you using? An SSD1306 based device? What controller are you using? A Pyboard?

If a font contains the digits 0-9 and alphabetic characters you should be able to display strings like "1900Kg".

fourlexboehm, 2017-03-09:
Yes I am using an ssd1306 on an esp8266 which has a very simple micropython driver that currently supports a small standard font size. I've experimented with other displays since but the contrast/brightness in industrial conditions isn't sufficient.
 I implemented a basic version using a driver in the arduino sdk but I have little experience with how I would create the wifi backend. In micropython I have a simple script to connect the esp to the wireless and a bash backend that copies all measurments from a txt file and adds them to a postgresql database. 

This offers superb redundancy cause if the with goes out of range, the numbers will still be added to the database. With very little microcontroller knowlege I don't want to do this in C

peterhinch, 2017-03-09:
I suggest you raise this in the [forum](http://forum.micropython.org/). There are a lot of helpful people in the forum with a great knowledge of WiFi application design. GitHub is intended for bug reporting and tracking.

closed 2018-08-14 by peterhinch