peterhinch/micropython-font-to-py #4: Remove/clean string from display, allow better positioning

BoKKeR, 2018-01-16:
Hey, I do like the project but if I am right there are few things that are missing. I don't see any way to remove the text from the screen. This way new text prints under the old one. before I could use 
`display.fill(0) `
but it does not affect the Writer. Another issue I am facing is actually counting out the text size. For example, putting text in the corner should be done as 
`font.getsize("string")`
 and this could be divided from the width of the display thus placing the text in the corner no matter how long the string gets.

peterhinch, 2018-01-17:
The `Writer` class is minimal by design. From my point of view the project's aim was to provide a means of storing fonts in flash memory with a minimal RAM footprint and to provide an open-source means of producing the files. The `Writer` was intended as a minimal working proof of concept.

A realistic font rendering class capable of working with `framebuf` would be a great contribution. I'd rather keep `Writer` as a simple demonstrator but an improved version could be subclassed from it or developed independently. Another contributor attempted this, including such features as rotation. Unfortunately he wanted to change the font file format. I'd need  an extremely good reason to do this: there are existing drivers which use the format with their own rendering code - i.e. not using `Writer`.

Feel free to contribute or to develop it in your own repo. I'll happily link to any working implementation.

BoKKeR, 2018-01-31:
I managed to get these things working but I wont be putting it on GitHub as the solution is rather ugly

peterhinch, 2018-01-31:
Fair enough. I'll close this for now.

closed 2018-01-31 by peterhinch