peterhinch/micropython-font-to-py #7: usable characters in conversions

diginfo, 2018-05-20:
Hi Peter

Apologies for bothering you with this, but have been working on displays & fonts intensively and as a result are coming up with some issues.

I am attempting to convert icon fonts, but as many of these use unicode (> 255) I have created new fonts by splitting the file into several bits.

This works, however even though I am using chars 1 - 255 in my conversion ( -s 1 -l 255 ) , when printing to the display I can only access the first glyph/icon (1) as chr(34)  and the last character I can access is chr(222) 

I am not sure if this is related to the conversion or to python itself - any ideas ??

**This is the fontmap showing 1 - 255:**

[image]

**And this is the generated file (1 of 3)**
[fa32_1.py.zip](https://github.com/peterhinch/micropython-font-to-py/files/2020006/fa32_1.py.zip)

peterhinch, 2018-05-20:
I've never tried an icon font but it should work. Alas I don't entirely follow the problem. Using your file I did the following:
```python
>>> m =fa32_1.get_ch(chr(0))
>>> m
(<memoryview>, 32, 28)
>>> bytearray(m[0])
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00\x0e\x00\x00\x00\x1f\x00\x00\x00?\x80\x00\x00\x7f\x00\x00\x00\xfe\x00\x00\x01\xfc\x00\x00\x03\xf8\x00\x00\x07\xf0\x00\x00\x0f\xe0\x00\x00\x1f\xc0\x00\x00?\xff\xff\xe0\x7f\xff\xff\xf0\xff\xff\xff\xf0\xff\xff\xff\xf0\x7f\xff\xff\xf0?\x80\x00\x00\x1f\xc0\x00\x00\x0f\xe0\x00\x00\x07\xf0\x00\x00\x03\xf8\x00\x00\x01\xfc\x00\x00\x00\xfe\x00\x00\x00\x7f\x00\x00\x00?\x00\x00\x00\x1f\x00\x00\x00\x0e\x00\x00\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>> m =fa32_1.get_ch(chr(1))
>>> bytearray(m[0])
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\x81\x80\xff\x80\x00\x00\x00\x00\x00\x00\x00\x00')
>>> m
(<memoryview>, 32, 12)
>>> 
```
As far as I can see the first two characters can be accessed correctly as can a glyph with index 250:
```python
>>> m =fa32_1.get_ch(chr(250))
>>> m
(<memoryview>, 32, 32)
>>> bytearray(m[0])
bytearray(b'\x00\x00\x00\x1f\x00\x00\x01\xff\x00\x00\x1f\xff\x00\x00\x7f\xff\x00\x01\xff\xff\x00\x07\xff\xfe\x00\x1f\xff\xfe\x00?\xff\xfe\x00\x7f\xff\xfe\x00\xff\xff\xfc\x01\xff\xff\xfc\x01\xff\xff\xfc\x03\xfc\xff\xf8\x03\xf8\xff\xf8\x07\xf1\xff\xf8\x07\xe3\xff\xf0\x07\xc7\xff\xf0\x0f\x8f\xff\xe0\x0f\x1f\xff\x00\x0e?\xff\x00\x0c\x7f\xff\x00\x08\xff\xff\x80\x01\xff\xff\x00\x03\xff\xfe\x00\x07\xff\x00\x00\x0f\xff\x00\x00\x1f\xff\x00\x00?\xfc\x00\x00|\x00\x00\x00\xf8\x00\x00\x00\xf0\x00\x00\x00`\x00\x00\x00')
>>> 
```
Are you sure your method of creating the index to access the glyph is correct? You might want to print `ord(index)` to verify this.

diginfo, 2018-05-20:
Seems to be some problem with possibly the display and chars < 32.

I just finished re-ordering the icons starting from the space char (32) and now it seems to be working as expected on the display:

```
def faat(x,y,cid,idx=1,invert=False):
  oset = 31
  if idx==1:  ## 32 - 224
    from fonts import fa321
    fidx = fa321
  elif idx==2:  ## 32 - 224 
    from fonts import fa322
    fidx = fa322
  elif idx==3:  ## 32 - 180 
    from fonts import fa323
    fidx = fa323
  Writer.set_textpos(y,x)
  Writer(e,fidx).printstring(chr(cid+oset),invert=invert) 
```

Here's the files if you would like to try:

[fa.zip](https://github.com/peterhinch/micropython-font-to-py/files/2020123/fa.zip)

diginfo, 2018-05-20:
Hi again Peter;

The attached file reports a max width of 242 pixels:

```
font2py -s 32 -l 210 -x ./fa-3of3a.ttf 32 ./fa323.py
Writing Python font file.
Height set in 2 passes. Actual height 32 pixels.
Max character width 242 pixels.
./fa323.py written successfully.
```

[fa-3of3.ttf.zip](https://github.com/peterhinch/micropython-font-to-py/files/2020265/fa-3of3.ttf.zip)

peterhinch, 2018-05-21:
That file contains glyphs of hugely variable heights and widths. While I haven't measured them a width of 242 does indeed look inconsistent with a height of 32.

The utility takes a set of glyphs whose height varies and maps them onto rectangular spaces which have variable width but the same height. Even if it were to work with this file the outcome wouldn't be good because the smallest glyph would occupy an inordinately large vertical space.

Frankly I'm amazed **font2py** works at all with such data. It was designed to convert character fonts and has only been tested with such. I suggest you restrict it to files whose icons have a constant height. It would not be practical to adapt font2py to cope with files like **fa-3of3.ttf** because the format of the **.py** file is based on a constant glyph height.

closed 2018-08-14 by peterhinch