peterhinch/micropython-font-to-py #12: sh1106 compatability

pacmac, 2018-11-15:
Hi Peter;

https://github.com/robert-hh/SH1106

Am trying to use the writer with the sh1106 but I get:

`ValueError: Device must be derived from FrameBuffer.`

But it looks like this is derived from framebuffer ??

```
import sh1106
>>> ssd = sh1106.SH1106_I2C(128, 64, gpio.i2c)
>>> dir(ssd)
['__class__', '__dict__', '__init__', '__module__', '__qualname__', 'addr', 'blit', 'buffer', 'fill', 'fill_rect', 'framebuf', 'hline', 'invert', 'line', 'pixel', 'rect', 'reset', 'scroll', 'sleep', 'text', 'vline', 'res', 'init_display', 'poweroff', 'poweron', 'contrast', 'show', 'width', 'height', 'external_vcc', 'pages', 'write_cmd', 'write_data', 'i2c', 'temp', 'rotate', 'hw_write_data', 'sw_write_data']
```

pacmac, 2018-11-16:
I managed to successfully convert the driver to inheritance:

https://github.com/pacmac/micropython-share/blob/master/sh1106_i2c.py

peterhinch, 2018-11-16:
I suggest you submit a PR: I'm sure Robert will appreciate it. Subclassing `framebuf` offers the benefit of providing the `framebuf` graphics primitives at zero cost.

closed 2018-11-16 by peterhinch