peterhinch/micropython-font-to-py #15: Fremebuffer Inheritance

pacmac, 2019-04-12:
Hi Peter;

Just tried to get some old epaper code working and are having trouble with the writer class:

```
MicroPython v1.9.3-1740-g4fdc731ef-dirty on 2019-04-12; PYBD_SF2W with STM32F722IEK
Type "help()" for more information.
>>> from writer import Writer as w
>>> w.set_textpos(0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "writer.py", line 55, in set_textpos
  File "writer.py", line 45, in _get_id
ValueError: Device must be derived from FrameBuffer.
```

peterhinch, 2019-04-12:
What device driver are you using? As stated in WRITER.md
> These classes facilitate rendering Python font files to displays where the
display driver is subclassed from the `framebuf` class.

It looks as if your device driver does not meet this criterion.

pacmac, 2019-04-13:
Apologies, I was calling the writer wrongly.

closed 2019-04-13 by pacmac