peterhinch/micropython-font-to-py #17: "Device not derived from FrameBuffer" using a font from font_to_py in custom firmware

xonu314, 2019-10-04:
Hello, 

first of all, thank you very much for providing these utilities, enabling to use custom fonts on an OLED with Micropython. 
 
I have an OLED (128x64) connected to an ESP8266. Using the SSD1306 driver with the latest Micropython firmware (esp8266-20190529-v1.11.bin), I am able to use various fonts after "converting" them with your font_to_py script and to show them on the display using writer.py. 
I have some sensors (DHT22 and a light dependent resistor) connected to the ESP. I read the values and display the results on the OLED. Initially all works well. However, sooner or later I get following error: 
 `MemoryError: memory allocation failed, allocating ... bytes`

According to your comments from #11 I have compiled a custom firmware, in which I have included my font file (generated from font_to_py). I built the new firmware following this tutorial: https://www.youtube.com/watch?v=jG7WBY_vmpE

Using the custom firmware I do the following:
```
from time import sleep
from machine import Pin, I2C
import ssd1306
from writer import Writer
i2c = I2C(-1, scl=Pin(5), sda=Pin(4))
oled_width = 128
oled_height = 64
oled = ssd1306.SSD1306_I2C(oled_width, oled_height, i2c)
wri = Writer(oled, monteserrat16)
```

The last line causes:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "writer.py", line 39, in __init__
  File "writer.py", line 14, in _get_id
ValueError: Device must be derived from FrameBuffer.
```

Using the official firmware, the exact same lines do not cause any error and I am able to display text using 

```
wri.printstring('Hello World!')
oled.show()
```
With the custom firmware I can show text with:
```
oled.text('Hello World!', 0, 0)
oled.show()
```
Following are the modules in my custom firmware:

```
>>>help('modules')
_main__          inisetup          socket            upip_utarfile
_boot             io                ssl               urandom
_onewire          json              struct            ure
_webrepl          lwip              sys               uselect
apa102            machine           time              usocket
array             math              ubinascii         ussl
binascii          micropython       ucollections      ustruct
btree             monteserrat16     ucryptolib        utime
builtins          neopixel          uctypes           utimeq
collections       network           uerrno            uwebsocket
dht               ntptime           uhashlib          uzlib
ds18x20           ntptimecustom     uheapq            verdana14
errno             onewire           uio               webrepl
esp               os                ujson             webrepl_setup
flashbdev         port_diag         umqtt/robust      websocket_helper
framebuf          random            umqtt/simple      zlib
gc                re                uos
hashlib           select            upip
```

Do you have any idea about what I am not doing correctly? 

I don't have much experience with Micropython or ESPs and topics such as Framebuf are still "black boxes" to me, so if I am not seeing something obvious or if I am posting this at the wrong place, I do apologize. 

Best regards, 
Sonu

peterhinch, 2019-10-05:
You are using an old version of ssd1306.py. Go into the directory holding the copy of the MicroPython repo on your PC and issue
```
$ git pull origin master
```
This will update your sourcecode. You will find the new version in drivers/display - [this line](https://github.com/micropython/micropython/blob/27fe84e661d664c5c145b4b3a8d544dad7de0acb/drivers/display/ssd1306.py#L28) shows the feature whereby it is derived from framebuf.

I suggest copying the file to ports/esp8266/modules and rebuilding. That way it will be frozen as bytecode and will save RAM.

xonu314, 2019-10-05:
Thank you very much. You were right, I was indeed using an old version of ssd1306.py. Seems obvious now that you have mentioned it ;-) Using the new version in my modules directory for my custom firmware, I can now display custom fonts and so far I am not seeing any memory issues (although it is only running for about an hour or so). Thanks for your great support!

closed 2019-10-06 by peterhinch