peterhinch/micropython-font-to-py #19: Different sized fonts on the same line do not line up properly

ironss, 2019-10-17:
The bitmaps are drawn from the top of the bitmap. This means text on the same line but with different sized fonts does not line up correctly. To line up nicely, the baseline of the fonts should line up, not the top of the font.

Of course, the original font knows the offset of the top of each bitmap from the baseline, and the generating code knows it too, as 'max_ascent'. However, this is not carried through into the generated font.py file.

A simple change (https://github.com/peterhinch/micropython-font-to-py/commit/c1085f28d1b2da4c0744180b25c141121aeeefe4) creates a new function, baseline() that returns the position of the baseline from the top of the font bitmaps.

peterhinch, 2019-10-17:
This looks good and I'll push an update shortly. Please could you provide a code sample of how this should be used so I can update the docs.

peterhinch, 2019-10-18:
I have pushed an update.
I take it that the baseline is the offset of a notional "ruler line" relative to the top. A ruler line being analogous to the lines on ruled paper where you might write with descenders lying below the line; different fonts have the same relationship to that line.

peterhinch, 2019-10-19:
OK, I found [this explanation](https://stackoverflow.com/questions/27631736/meaning-of-top-ascent-baseline-descent-bottom-and-leading-in-androids-font) so closing this.

Thanks for your work on these issues.

closed 2019-10-19 by peterhinch