peterhinch/micropython-font-to-py #26: Bitrot in font_test.py?

daniel-thompson, 2020-02-13:
Hi Peter

Seems like I've ended up touring a lot of your projects recently ;-)

I've only really been kicking the tyres so far but I did notice that several of the Font() instantiations in font_test.py don't supply enough arguments for the tests to work correctly. It definitely affects font_test() and I think it would also impact test_stream() and test_arrays() too.

Locally I have added some default arguments to the Font() constructor to help work round it but wasn't sure whether default arguments or fixing the tests is a better (so no PR for now).

Daniel.

peterhinch, 2020-02-14:
Good point, thank you for raising this. In practice the only test I actually use is **test_font** which has proved extremely useful. 

The other functions were used during early development of the code. Unless you have a particular reason for running them I may simply delete them.

daniel-thompson, 2020-02-14:
No, I have not particular need to run them. The only reason I was looking at them is because I started looking at the tests to try and figure out how it all hangs together. I have to dig relatively deep because I think I will need alternative Writer() classes; my display is larger than RAM so it does not use the framebuf abstraction.

peterhinch, 2020-02-14:
I have pushed an update: `font_test.py` is now stripped down and is documented. It can now be run at the command prompt. The remaining functionality is the only part I've used in a long time. I have found it useful, to check fonts and in testing when people have submitted patches to `font_to_py`.

I have solutions for various displays with internal frame buffers. See for example this in my [RA8875 driver](https://github.com/peterhinch/micropython_ra8875/blob/76608da372e5e1ff089f4628860abfb209d2c487/driver/ra8875.py#L326).

daniel-thompson, 2020-02-17:
Thanks for the RA8875 hint. I'll take a look at that.

closed 2020-02-17 by daniel-thompson