peterhinch/micropython-font-to-py #27: Not able to create font

pynadjim, 2020-02-25:
Hi I am not able to create font using the font_to_py.py. I want to create freesans24.py and freesans36.py

peterhinch, 2020-02-26:
It would help if you explained the nature of your difficulty. This works here:
```bash
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython-font-to-py
$ ./font_to_py.py -x FreeSans.otf 24 freesans24.py
Writing Python font file.
Height set in 2 passes. Actual height 24 pixels.
Max character width 25 pixels.
freesans24.py written successfully.
```

Down10, 2020-05-12:
I was having problems with this same issue myself, but the script worked once I changed the first line of _font_to_py.py_ (and _font_test.py_) from this:
`#! /usr/bin/python3`
to this:
`#!/usr/bin/env python3`

So perhaps those files ought to be updated, to allow for alternate Python install environments.

peterhinch, 2020-05-12:
Thanks for that. Now done.

closed 2020-05-12 by peterhinch