peterhinch/micropython-font-to-py #33: Writer hangs if font.height larger than device.height

Ian9231, 2021-02-03:
I had modified the SSD1306 driver to have two frame buffers (display is 128 x 64 with the top 16 row yellow and remainder blue). the upper buffer is 128 x 16 and the lower 128 x 48. Writer works fine on the lower buffer and can be instantiated on the upper using the freesans10.py font. Writer then hangs when printstring is called.  If row clipping is set True the printstring returns but no text is displayed.  
Note freesans10.py is 17 pixels high.

My work around in __init_ is
self.row_clip = (self.font.height() > self.screenheight)
to prevent hanging.

peterhinch, 2021-02-04:
That is an extreme case. Maybe the constructor should throw an exception: such a Writer really isn't much practical use.

Ian9231, 2021-02-04:
True, an exception would be better, as it was not initialy apparent to me that the freesans10.py font was 17 height. And my work around results in no text being displayed.  Sent from Samsung tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/peterhinch/micropython-font-to-py/issues/33#issuecomment-773188815",
"url": "https://github.com/peterhinch/micropython-font-to-py/issues/33#issuecomment-773188815",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

peterhinch, 2021-02-06:
Update which throws a **ValueError** now pushed.

closed 2021-02-06 by peterhinch