peterhinch/micropython-font-to-py #36: Scaling font

varna9000, 2021-09-05:
Hi Peter,
I've been using this tool for more than year and my biggest concern is the size of the output font file. I'm generating mainly Cyrillic fonts and I've noticed two things - the bigger the font size, the bigger the .py file. Also, if I generate same amount of chars latin letters and same font size, the .py file is smaller. Does the char number in  utf-8 table somehow influence .py file size ( the higher the char number, the bigger the file size)?  Experiencing the above, eats up a lot of esp8266 memory, even if I compile it to .mpy

So I though there could be a workaround by adding a function to scale a small font e.x. x1.5 or x2 This would keep file size small and take less memory. Would that be easy to add?

peterhinch, 2021-09-05:
The solution is to use frozen bytecode, which is better than compiling to MPY. Fonts then use Flash space but almost zero RAM.

varna9000, 2021-09-06:
Thanks, Peter. I'll try this even with the other external modules I'm using in the project.

closed 2021-09-06 by varna9000