peterhinch/micropython-font-to-py #38: How to add the Euro sign

PaulskPt, 2022-01-04:
Hi Peter,
For a project using the following hardware: 
A ```Pimoroni Pico Breakout Garden base```, onto which stacked: a ```Raspberry Pi Pico```; a ```Pimoroni BG RV3028 rtc module```; an ```Adafruit TMP117 sensor``` via STEMMA/Qwiic connection wired to a ```Pimoroni BG adapter```; a ```Pimoroni BG 1.12in OLED SPI 128x128px monochrome display```.

I successfully created a font file using your ```font-to-py``` repo.
Using the -k parameter I added the extended characters (```python3 ./font_to_py.py -x -k extended NotoSansDispCondMed.ttf 18 my_extended18hor.py```).

For a future expansion of my project I want to be able to print the ```Euro sign```. I saw that the Euro sign does not exist in the file: ```extended```.

I tried to study the created font file. I see that the last part, a bytearray, with the name ```_sparse``` contains a certain pattern, consisting of 4 bytes starting with a value that appears to me to be a value into the Extended ascii chart (e.g. published on: ```https://www.ascii-code.com```). In the file 'extended' I replaced the ```Pound Sterling``` sign, the last character in that file, with the ```Euro sign```. Now the contents of the file 'extended' is:
``` ¬!"#£$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~°Ωαβγδθλμπωϕ€```

Then I built a new version of the font file using the -k parameter. However, in the test, when using a string value of ```chr(128)```  which is the ordinal value for the ```Euro sign``` in the ```Extended ascii``` chart (on ascii-code.com at least), still a question mark ('?') is printed in the REPL.
                                   
In the file: font_to_py.py, Class Font, I added the following line:
 ```289           print("len(cs): {}, ultimate char: {}, ord ultimate char: {}".format(len(cs), cs[len(cs)-1], ord(cs[len(cs)-1])),end='\n')```
The result of this addition in the output was:
                                   ```len(cs): 8333, ultimate char: €, ord ultimate char: 8364```  (see also the image below of a font building run output).

Analyzing the creation of the font file by inserting another line 289 (printing the value of ```cs```) to the function ```__init__()``` of the Class ```Font``` in file: ```font_to_py.py```, revealed that the Pound Sterling sign still was present in the created font file.

When I changed the contents of the file ```extended``` in such a way that I put the ```Euro sign``` at it's ascii ordinal place,
that is: between ```DEL``` and the ```degrees sign```, as shown here:

[image]

then the result of building the font file is that the  ```Euro sign``` (again) is moved to the end of the string, as showed in the build output:

[image]

I copied various functions from the file ```font_test.py``` into my ```main.py``` script.
In the definition of the function ```test_font()```, when I changed the default value for parameter ```string``` into:
      
```
def test_font(fontfile, string=''+'abg'+chr(176)+chr(163)+chr(8364)+chr(98)+chr(110)+chr(120)+chr(131)+chr(181)):
                                          |          |         |
                                        degs (°)     £         €
```
while, for the ```Euro sign```, using the ord() value of 8364 as indicated by the ```font_to_py.py``` script,
the result in REPL is that the ```degrees sign``` and the ```pound sterling sign``` are printed however a ```'?'``` was printed for the ```Euro sign```.

I guess that the ```Euro sign``` is not defined inside the created font file: ```my_extended18hor.py```.

```Question:```
How can I get the wanted output for the Euro sign being printed in REPL and on the OLED display?

Paulus Schulinck

Attachments: 1) REPL output during run of my script; 2) image of the hardware.

[image]

[image]

peterhinch, 2022-01-05:
Have you checked that your source `otf` or `ttf` font includes a Euro glyph?

PaulskPt, 2022-01-05:
I used a ttf font that I created from Google fonts service.  Below a cut of the ```Create_font.pde``` that I used to create the font. This font I used for another project.

[image]

PaulskPt, 2022-01-05:
I just created a glyphs test function. In the (partly) print of the run result it shows that there are no glyphs for in the range: 128 - 160. It starts with 161 (and later on a lot of other glyphs among them arabic. I guess I have to create another .ttf. I come back when I have done that.

[image]

PaulskPt, 2022-01-05:
I downloaded from Google Fonts: Noto Sans - Regular. Ran it with font-to-py. Imported the file ```my_notosansregluar18```.
Ran the test_glyphs() in range 8300-8365 and the Euro symbol was printed. 

[image]

Then (just for test/fun) added the Euro symbol instead of  'C' (Celsius). See the result below. The Euro symbol was also displayed on the OLED.

[image]

Thank you for your answer / 'trigger'. Issue can be closed.

closed 2022-01-05 by PaulskPt