peterhinch/micropython-font-to-py #39: ValueError: Font must be horizontally mapped.

wumfi, 2022-01-16:
As per issue #10, I too am getting the above error with any ttf I find online.

Is there something specific I should be searching for to find horizontally mapped fonts?

Each file that I run through `font_to_py.py` produces a file where the hmap function is returning False.

peterhinch, 2022-01-17:
You need to run `font_to_py` with the `-x` argument to produce a horizontally mapped Python font.

Run
```bash
$ font_to_py --help
```
for a list of args.

closed 2022-01-17 by peterhinch

easytarget, 2024-03-11:
This should be **properly** and **prominently** documented.
The _default_ for your tool is that it produces results that are incompatible with the example Writer class provided. I'm agog.

peterhinch, 2024-03-13:
The correct mapping depends on your display hardware. Vertically mapped fonts are compatible with `Writer` and display correctly on hardware which uses vertical mapping.

lerouxb, 2024-03-21:
> The correct mapping depends on your display hardware. Vertically mapped fonts are compatible with Writer and display correctly on hardware which uses vertical mapping.

Is this true? Reading the code there's if font.hmap() and then in else it throws. Immediately after that it has a hardcoded print that stats that the orientation is horizontal: https://github.com/peterhinch/micropython-font-to-py/blob/c37e4acabd2e43eaae468c0f1854f35be86c27bd/writer/writer.py#L72-L77

peterhinch, 2024-03-22:
Apologies, you are right. A hazard of old age - you forget how your own code actually works...

The reason for the `-x` flag is historical. I started work on `font-to-py` nearly ten years ago for an early (now obsolete) e-paper display which was vertically mapped. I therefore made vertical mapping the default. (Note that `framebuf` has some support for vertical mapping). Alas horizontal mapping seems to be universal in modern displays.

While it would make sense to make horizontal the defualt in `font-to-py` - and also to scrap binary fonts, another relic - I'm reluctant to make breaking changes. There is always someone who pops up, howling in anguish.

Perhaps the way forward is for `font_to_py.py` to issue a warning message if a font is created without the `-x` arg?

lerouxb, 2024-03-22:
No need to apologize. I'm much younger and have to find a team member to explain to me my own diffs from weeks ago on a surprisingly regular basis :)

I think maybe just a notice / paragraph added to the markdown file should be fine.

A side-note: I misinterpreted the phrasing there for some reason. I couldn't tell if the flag meant "This .bdf file IS horizontally mapped" or "map this horizontally when writing the .py file" and went on a detour looking into the specs of how the files work 😆

peterhinch, 2024-03-22:
I have now pushed an update. The default mapping is horizontal. The `-x` arg still exists but has no effect. There is a new `-y` or `--ymap` arg that specifies vertical mapping.

The docs further clarify that mappings refer to the output file and display hardware. They also state that vertical mapping is incompatible with `Writer` classes.

easytarget, 2024-05-01:
Sorry for shouting in my earlier comment, and thank you for fixing this.
.. I should probably apologise for necro-posting a closed issue too ;-)