peterhinch/micropython-font-to-py #40: Support for Waveshare: Pico LCD 1.14

wx4cb, 2022-01-19:
is there a display driver for the above board? it uses the ST7789VW chipset for the display. the Docs are below.

https://www.waveshare.com/wiki/Pico-LCD-1.14

I've managed to get the normal writer class working with not much issues, but as it's a colour display, i'd obviously like to use colours on the display.

any help would be appreciated.

wx4cb, 2022-01-19:
update. i found there's an SS7789 driver in the nano_gui repo:

https://github.com/peterhinch/micropython-nano-gui/blob/master/drivers/st7789/st7789_4bit.py

but i can't get it running.... i've set the chip selects etc and tried copying the init lines from the "official" sample code to init the SPI, but it doesnt work....

here's the code im working with right now, taken from the writer.md page

```
from  machine import SPI, Pin
import gc
import time
from writer import CWriter
import freesans20  # Font to use
from drivers.st7789.st7789 import ST7789 as SSD  # Adapt for other hardware

# Needed on my Pyboard D PCB to enable supply to the display
#pp = machine.Pin('EN_3V3')
#pp(1)
#time.sleep(1)

# Adafruit options
# height = 96  # 1.27 inch 96*128 (rows*cols) display
height = 135 # 1.5 inch 128*128 display
width = 240

BL = 13
DC = 8
RST = 12
MOSI = 11
SCK = 10
CS = 9
    
pdc = Pin(DC, Pin.OUT, value=0)
pcs = Pin(CS, Pin.OUT, value=1)
prst = Pin(RST, Pin.OUT, value=1)
#spi = SPI(1)
#spi = SPI(1,1000_000)
spi = SPI(1,10000_000,polarity=0, phase=0,sck=Pin(SCK),mosi=Pin(MOSI),miso=None)

gc.collect()  # Precaution before instantiating framebuf
ssd = SSD(spi, pcs, pdc, prst, height, width)  # Create a display instance

# Define a few colors
GREEN = SSD.rgb(0, 255, 0)
RED = SSD.rgb(255,0,0)
BLACK = SSD.rgb(0, 0, 0)

# Instantiate a writer for a specific font
wri = CWriter(ssd, freesans20)  # Can set verbose = False to suppress console output
CWriter.set_textpos(ssd, 10, 10)  # In case a previous test has altered this
wri.setcolor(RED, BLACK)  # Colors can be set in constructor or changed dynamically
wri.printstring('Sunday\n12 Aug 2018\n10.30am')
ssd.show()

while True:
    ...
```

peterhinch, 2022-01-20:
If you look at the [setup example for this display](https://github.com/peterhinch/micropython-nano-gui/blob/master/setup_examples/st7789_pico_lcd_114.py) you will see that the SSD constructor needs a `display=TDISPLAY` arg:
```python
from drivers.st7789.st7789_4bit import *
# code omitted
ssd = SSD(spi, height=135, width=240, dc=pdc, cs=pcs, rst=prst, disp_mode=LANDSCAPE, display=TDISPLAY)
```
I have tested these displays with this driver and [micro-gui](https://github.com/peterhinch/micropython-micro-gui) so it should work.