peterhinch/micropython-font-to-py #46: Writer: Unpredictable behaviour when text overlaps bottom edge of screen

sandyscott, 2023-01-19:
When text overlaps the bottom edge of the screen the "black" area extends to the full width of the screen (instead of just behind the text) and items placed before that `printstring()` call are ignored.

Tested with a Pi Pico talking to an 128x32 SSD1306 monochrome OLED display over I2C
Display: https://www.waveshare.com/wiki/0.91inch_OLED_Module

Using the code below, the picture shows two situations

- When `second_line_position = 14`, the second line just touches the bottom of the screen (32 pixel screen height, font height is 18px: 32 - 18 = 14). This is the output I expected - both lines visible, black area is only behind the text.
- When `second_line_position = 15`, the second line overlaps the bottom of the screen by 1 pixel: this results in the first line disappearing, and the black area behind the text filling the full width of the screen.

[image]

Demonstration code:
```python
from machine import Pin, I2C
from writer import Writer

# From https://github.com/micropython/micropython-lib/tree/master/micropython/drivers/display/ssd1306
import ssd1306
# uw-ttyp0, 18 px tall, bold. standard variants, converted from the bcf file.
import uw_ttyp0_18b

# Initialise I2C
i2c = I2C(0, sda=Pin(0), scl=Pin(1))
display = ssd1306.SSD1306_I2C(128, 32, i2c)

# Initialise display and fill with white
wri = Writer(display, uw_ttyp0_18b)
display.fill(1)

# First Line
wri.set_textpos(display, row=0, col=0)
wri.printstring('abcd')

# Second line
second_line_position = 14
wri.set_textpos(display, row=second_line_position, col=0)
wri.printstring('1234')

display.show()
```

sandyscott, 2023-01-19:
I should have said why I would want the text to overlap the bottom of the screen - with applications that don't need to use descenders - e.g. numbers only, you could fill the screen better with a bigger font.

peterhinch, 2023-01-20:
This is rather a special case. My GUI's adopt the approach of printing a warning if a text object overlaps any edge of the screen. While this wouldn't help with your case, you'll probably see where I'm coming from: overlaps should be detected at the point that a text object is instantiated rather than when a glyph is rendered. Further, the glyphs in a bitmapped font are all the same height, so it's not practicable to detect which glyphs have descenders at runtime. So I don't plan to fix this.

I think your best solution might be to investigate fonts. Are there any number-only fonts that are designed to lack descenders?

closed 2023-01-20 by peterhinch