peterhinch/micropython-font-to-py #47: official ssd 1306 driver

jpnielsen, 2023-05-16:
Hi,

Beginner here, trying to get micro-gui running on a raspberry pi pico + SSD1306 I2C display.

I grabbed micropython 1.20 for my pico - but it wouldn't import ssd1306

Your link to "original ssd1306 driver" on this page https://github.com/peterhinch/micropython-font-to-py/blob/master/writer/DRIVERS.md

gives a "404 - page not found"

i tried combining new and older stuff - but the mini gui simple demo kept complaining that my ssd did't have "rgb".

Hope you'll look into it, in case micropython moved something around.

jpnielsen, 2023-05-16:
I'm not 100% sure this is the file I'm looking for

https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/display/ssd1306/ssd1306.py

When I use it with "writer.py" it gives an error:

**  File "writer.py", line 50, in set_textpos
  File "writer.py", line 40, in _get_id
ValueError: Device must be derived from FrameBuffer.**

peterhinch, 2023-05-16:
The drivers in micropython-lib were moved - you have found the correct version. Thanks for the report: I have corrected the links.

I think your code must be importing an invalid version as the error message does not make sense: the official version is derived from `FrameBuffer` as a glance at the code will confirm. Please ensure that the only copy of `ssd1306.py` on your device is from [here](https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/display/ssd1306/ssd1306.py).