peterhinch/micropython-font-to-py #49: writer_demo.py

brylka787, 2023-08-31:
Hi Peter,
i am new to micropython on the pi pico.
I don't understand "writer_demo.py".
# Create a display instance
# verbose = False to suppress console output
and "ssd1306_setup.py"
else: # I2C
         # Pyb SSD
         #3v3 Vin
         # Gnd Gnd
         # Y9 CLK
         # Y10 DATA
         if soft:
             pscl = machine.Pin('Y9', machine.Pin.OPEN_DRAIN)
             psda = machine.Pin('Y10', machine.Pin.OPEN_DRAIN)
             i2c = machine.I2C(scl=pscl, sda=psda)
         otherwise:
             i2c = machine.I2C(2)
         ssd = SSD1306_I2C(WIDTH, HEIGHT, i2c)

I use I2C and what is if soft:
I can't get "writer_demo.py" to work.
Nothing on display.
in
https://github.com/peterhinch/micropython-font-to-py/tree/master/writer
Peter

peterhinch, 2023-09-02:
You'll need to edit `ssd1306_setup.py` to match the electrical connections to the SSD1306 display. Existing pin numbers are for the Pybord.

brylka787, 2023-09-02:
Hi Peter,
for the Raspberry Pi Pico I have
if soft:
             pscl = machine.Pin((1), machine.Pin.OPEN_DRAIN)
             psda = machine.Pin((0), machine.Pin.OPEN_DRAIN)
             i2c = machine.I2C(scl=pscl, sda=psda)
         otherwise:
registered.
but other users use:
i2c = I2C(0,scl=pin(1), sda=pin(0))

Can you use ssd1306_setup.py for the
Raspberry Pi Pico
not use at all

Peter

peterhinch, 2023-09-03:
Hardware I2C(0) on the Pico uses GPIO9 (SCL) and GPIO8 (SDA) - alternatively GPIO17 (SCL) and GPIO16 (SDA).
```python
from machine import Pin, I2C
i2c = I2C(0, scl = Pin(9), sda = Pin(8))
```
You need to specify the pins because the Pico has two alternative pinouts for I2C(0).

brylka787, 2023-09-03:
Hi Peter,

Thanks again for the info
with pull-up resistors.
and the multiple assignments of the pins for the I2c bus.
At the Pico.

I'm a Python beginner and was fascinated by the module

ssd1306_setup.py

with the transfer of the parameters for the definition from the program.

writer_demo.py

and return with return.

But I still don't understand what that means.

# verbose = False to suppress console output
in writer_demo.py

Is that from
if verbose:
             print('Orientation: {} Reversal: {}'.format('horiz' \
                 if font.hmap() else 'vert', font.reverse()))

from writer.py
and how to change.

Peter

peterhinch, 2023-09-04:
You pass the `verbose` flag as an optional third argument to the `Writer` constructor. To suppress console oututput, pass `False`:
```python
wri = Writer(ssd, freesans20, False)
```

brylka787, 2023-09-04:
Hi Peter,

Thanks,
Now I'm slowly beginning to understand
how to pass parameters to functions and modules.
And get it back too.

But I'm still looking for a solution to this
SSD1306 .96 inch display
the representation of cyrillic characters.
I can already use font_to_py.py -x -l255.
But which .TTF file can I take

Peter

peterhinch, 2023-09-05:
I'm not sure what the problem is. There is an example [in the docs](https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md#312-optional-arguments) for creating a font with Cyrillic characters. Clearly the TTF file must include these - to check this you could try using the TTF font on a PC before converting it.

brylka787, 2023-09-10:
Hi Peter,

that was a bit misleading.

I didn't know at first.
Where can I get a Cyrillic TTF file.
And I needed a Serbian character TTF.

But I found the TTF on the internet.

And I transferred the file to a .py with font_to_py.py -x.

I can now use the Serbian alphabet on the display.

Works flawlessly.

Thank you very much again for all the information.

greeting
Peter from Bamberg in Germany.

peterhinch, 2023-09-10:
I'm glad it's working. I'm afraid I'm not much help on non-English character sets - the input on those has come from other contributors.

closed 2023-09-10 by peterhinch

brylka787, 2023-10-08:
Hi Peter,

Thanks again for the advice back then.
All problems are solved.
But I have a new problem.
I have a new display:
1.77" TFT LCD SPI
and use the module for this:

from ili934xnew import ILI9341, color565
and
import tt14
import tt24

German umlauts are not included in tt14 and tt24.
But I need umlauts.

tt14 and tt24 are included
font_to_py.py
# Code generated by font-to-py.py.

# Font: CM Sans Serif 2012.ttf

version = '0.2'
created.
I got these from:
https://github.com/jeffmer/micropython-ili9341

my self-created fonts are with:
# Code generated by font_to_py.py.
# Font: arial.ttf
# Cmd: font_to_py.py -x -l255 arial.ttf 16 arial16255.py
version = '0.33'
created.

When I use mine, I get the following error with display.set_font(arial16255):

Traceback (most recent call last):
   File "<stdin>", line 60, in <module>
   File "ili934xnew.py", line 318, in print
AttributeError: 'module' object has no attribute 'get_width'

line 60 is display.print("Ä Ö Ü")
line 318 is if curx + self._font.get_width(word) >= self.width:

In my arial16255.py
is there end:

_mvfont = memoryview(_font)
_mvsp = memoryview(_sparse)
ifb = lambda l : l[0] | (l[1] << 8)

def bs(lst, val):
     whileTrue:
         m = (len(lst) & ~ 7) >> 1
         v = ifb(lst[m:])
         if v == val:
             return ifb(lst[m + 2:])
         if not m:
             return 0
         lst = lst[m:] if v < val else lst[:m]

def get_ch(ch):
     doff = bs(_mvsp, ord(ch)) << 3
     width = ifb(_mvfont[doff : ])

     next_offs = doff + 2 + ((width - 1)//8 + 1) * 16
     return _mvfont[doff + 2:next_offs], 16, width
 
So the get_width(s): from tt14.py is missing.

How can I create a working arial16255.py.

Peter

brylka787, 2023-10-08:
Hi Peter,

I was too impatient again. !!!

I have the:

font_to_py.py

from the website:

https://github.com/jeffmer/micropython-ili9341

taken and with

python font_to_py.py -l255 arial.ttf 16 arial_16255.py

created.
Important: without -x for the application

And I have German umlauts.

P.S. I will still test -c and -k.

Peter

peterhinch, 2023-10-09:
The version of `font_to_py.py` on that website is **six years** old. A lot has been done in that time :)

You might also be interested in [this ili9341 driver](https://github.com/peterhinch/micropython-nano-gui/blob/master/drivers/ili93xx/ili9341.py): with that you can run nano-gui or micro-gui on these displays. Note that my recommendation of `-x` applies only to my display drivers. Others may well be written for vertical mapping.