peterhinch/micropython-font-to-py #58: Freely write at different coordinate

hunter40415, 2024-01-07:
I suggest add this line into the code to avoid having to call out a coordinate every time which make the code look messy
[image]
which turn the code from this
[image]
to this
[image]

peterhinch, 2024-01-08:
The reason `set_textpos` is a separate method is that, having set the text position, the application might perform many `printstring` calls with the text being added each time at the current cursor point. In other words, `printstring` should not be constrained to an absolute location.

closed 2024-01-08 by peterhinch

hunter40415, 2024-01-09:
thanks for your explanation. I still a amateur at coding so I haven't
thought about that. Anyway thanks to your work I have done many cool thing
on my pico pi.
(Mind my bad English vocabulary).

peterhinch, 2024-01-09:
You can always write your own function:
```python
def print_at(wri, text, row, col):
    Writer.set_textpos(row, col)
    wri.printstring(text)
```
then you can write
```python
wri = Writer(display, roman20)
print_at(wri, "Hello world", 0, 5)
```
I notice you use `x` and `y`. Graphics systems usually use `row`, `col`. The mathematical `x` coordinate is `col`, `y` is `row`. Just to confuse us :)