peterhinch/micropython-font-to-py #60: Color of txt

d3monlalov, 2024-04-03:
Hi! Im trying to use your driver for TFT ST7735 LCD and ESR32-Wroom. All right, but i dont understand how I can to change a color of text in  printstring function
My code below
`from machine import Pin, I2C,SPI
from ina219 import INA219
from logging import INFO
from ST7735 import TFT
from font import terminalfont
import time
from framebuf import FrameBuffer, RGB565
from writer import Writer
import freesans20
import courier20

buf = bytearray(160*128*2)
fb = FrameBuffer(buf, 160, 128, RGB565)

spi = SPI(2, baudrate=30000000, polarity=0, phase=0, sck=Pin(18), mosi=Pin(23), miso=Pin(12))
tft=TFT(spi,2,4,5)
tft.initr()
tft.rgb(True)
tft._setwindowloc((0,0),(159,127))
tft.rotation(1)
tft.fill(TFT.BLACK)

i2c = I2C(1, scl = Pin(22), sda = Pin(21))

def init_ina():
    SHUNT_OHMS = 0.1
    global ina
    ina = INA219(SHUNT_OHMS, i2c)#, log_level=INFO)
    ina.configure()
    
def i2c_scan():
    
    print('Scan i2c bus...')
    devices = i2c.scan()
    if len(devices) == 0:
      print("No i2c device !")
    else:
      print('i2c devices found:',len(devices))
    for device in devices:  
        print("Decimal address: ",device," | Hexa address: ",hex(device))

init_ina()
i2c_scan()
old_voltage = 0
old_time = 0

class NotionalDisplay(FrameBuffer):
    def __init__(self, width, height, buffer):
        self.width = width
        self.height = height
        self.buffer = buffer
        self.mode = RGB565
        super().__init__(self.buffer, self.width, self.height, self.mode)

    def show(self):
        ...

my_display = NotionalDisplay(160, 128, fb)
wri = Writer(my_display, freesans20)

 
while True:
    new_time = time.ticks_ms()
    new_voltage = ina.voltage()
    if (new_time - old_time) >1000 or abs(new_voltage-old_voltage)>0.05:
        fb.fill(0x0000)
        fgcolor = 56
        Writer.set_textpos(my_display, 20, 10)
        Writer.setcolor()
        wri.printstring("{0:.3f}".format(new_voltage))
        #fb.text(("{0:.3f}".format(new_voltage)),50,50,TFT.BLUE)
        tft._writedata(buf)
        old_voltage = new_voltage
        old_time = new_time
        `

peterhinch, 2024-04-04:
For color you need to use the `CWriter` class - `Writer` is for monochrome displays. `CWriter` is documented [here](https://github.com/peterhinch/micropython-font-to-py/blob/master/writer/WRITER.md#22-the-cwriter-class) with a usage example. 

An alternative approach is to use [nan-gui](https://github.com/peterhinch/micropython-nano-gui) which uses a `Label` class to provide for text objects whose contents and colors may be dynamically updated..

d3monlalov, 2024-04-06:
Thanks, all ok with СWriter, I have a good result!!! Some problems left with setup color because it value is not answer RGB565 format, but it's not so important for me

d3monlalov, 2024-04-06:
> For color you need to use the `CWriter` class - `Writer` is for monochrome displays. `CWriter` is documented [...]

Im just read documentation about CWriter and begin understand what I did wrong with color settings, thanks

closed 2024-04-07 by peterhinch