peterhinch/micropython-font-to-py #63: support hungarian characters

tilhoft, 2024-08-26:
It would be nice if I can use special chars (áíűüőöúé)
i believe it is utf-8, and i tried with an otf file which allegedly support hungarian chars

peterhinch, 2024-08-27:
Please see docs [para 3.1.2](https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md#312-optional-arguments). Support for special character sets is provided.

Witawat, 2024-10-27:
> Please see docs [para 3.1.2](https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md#312-optional-arguments). Support for special character sets is provided.

-k how to find ?

i try -k Thai or -k Thailand 

python .\font_to_py.py .\source\THSarabunNew.ttf 20 -x -k Thailand font_20.py   
Can't open Thailand for reading.

peterhinch, 2024-10-27:
You have to create the file `Thailand`. This must contain the glyphs you want to create. I created a file `thai` containing a copy of the glyphs in your other post. I then did:
```bash
[adminpete@meerkat]: ~/Downloads
$ font_to_py.py THSarabunNew.ttf 20 -k thai thailand.py
Writing Python font file.
Height set in 3 passes. Actual height 20 pixels.
Max character width 12 pixels.
Sparse font file.
thailand.py written successfully.
```

closed 2024-10-27 by peterhinch

peterhinch, 2024-10-28:
I have now posted the file `charsets/Thailand`. Thank you for the contribution.