peterhinch/micropython-font-to-py #64: 5x7 hand made fonts that work with writer.py

bradcar, 2024-10-17:
Hi Peter, Really like your font_to_py.py and our writer. You mentioned _"For small fonts it is best to use hand-designed bitmapped font files. "_ I want to use font's generated by your code as well as a 5x7 for numbers letters. where would I find a font_5x7.py that is compatible with yours?   Been looking but can't find anything yet.

bradcar, 2024-10-17:
Finally found one at: https://github.com/easytarget/microPyEZfonts/blob/main/examples/fonts/ezFBfont_07_font_tiny5_ascii.py

Might want to put reference to ez's fonts in your small font section of the readme, as others also may want to use both your work on writer, font_to_py, and larger fonts.  As well as smaller one.  THANKS AGAIN!

closed 2024-10-17 by bradcar

peterhinch, 2024-10-17:
Thanks for the pointer. README.md now [updated](https://github.com/peterhinch/micropython-font-to-py/blob/master/README.md#31-small-fonts).