peterhinch/micropython-font-to-py #66: How to convert Unicode characters

PythonKai, 2024-11-05:
Hello Peter, I found you font-2-py tool. I use it with sh1106 driver. 
Thank you very much for the great work. Unfortunately I’m not so familiar with TTF font format. 
Under Windows a character map (in the font folder) is findable, that shows a lot of Unicode characters.
I.e. In Segoe UI Symbol font (seguisym.ttf) the U-26E3 ( ☾ half-moon) character. 
Is it possible to convert the Unicode characters with you tool to py font ?
Can you help, and show how ?
Thank you very much 
BR Kai

peterhinch, 2024-11-05:
Perhaps I need to expand the docs as you're not the first to be puzzled by this.

See [optional arguments](https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md#312-optional-arguments). The one you want is `-k`. Create a file containing the entire character set you want to include. There are some examples in the `charsets` directory. Then you can use that charset file to create one or more fonts containing that character set. The doc has some usage examples for `-k`.

PythonKai, 2024-11-05:
I recognized the `-k`  parameter. Thank you for the advise.
Unfortunately  I don't know the `charsets ` in the `Segoe UI Symbol font (seguisym.ttf)` 
Do you know, where I will find the information ?
If the `charsets ` are known, how to catch the characters like `U-26E3` ? 
I work with Windows10 in the `cmd window` it is impossible to insert the unicode characters.
Is there a tip ?

peterhinch, 2024-11-07:
I abandoned Windows 15 years ago so I'm afraid you'll need to seek help elsewhere. A quick websearch yielded [this doc](https://support.microsoft.com/en-us/office/insert-ascii-or-unicode-latin-based-symbols-and-characters-d13f58d3-7bcb-44a7-a4d5-972ee12e50e0#bm1).

Nor do I know how to determine which glyphs a TTF font contains, other than trial and error.

The way to generate a charsets file is in a text editor. It's easiest to start with an existing file such as `charsets/extended` and simply add the glyphs you require (optionally removing any that are of no interest). Glyphs can be cut and pasted from another source, or there may be a set of keystrokes that produces the glyph.

This can be done using any font, but if you can set the editor to use the font you're targeting it will be apparent if a glyph you wish to create is not present in the font.

PythonKai, 2024-11-09:
Thanks for the hint. I found an online font editor and could copy the needed glyphs to a new ttf fontfile. That file was convered perfect by your `font-to-py` tool.

closed 2024-11-09 by PythonKai