peterhinch/micropython-font-to-py #67: Hebrew char set

regevt, 2024-12-05:
abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789!"#$%&'()*+,-./กאבגדהוזחטיכךלמםנסעפףצץקרשת 

also will require a rtl string function

```    
def printRTLstring(self, string, invert=False):
        reversed_string = ""
        for i in range(len(string) - 1, -1, -1):
            reversed_string += string[i]
        self.printstring(reversed_string, invert)
```

tested on a [Waveshare  Pico e-Paper 5.65](https://www.waveshare.com/wiki/Pico-ePaper-5.65)

peterhinch, 2024-12-06:
Thank you for the contribution. I've pushed the charset file.

I'll give some thought to RTL printing. The GUI `Label` classes can do right justification but I would like to devise a more efficient RTL algorithm if possible.

I think RTL management will be limited to single line `Label` instances. Widgets such as `Textbox` that allows scrolling text would be hard to adapt.

regevt, 2024-12-06:
I usually don't develop in Python, so it is possible that my code is not good.
The OG way of flipping a string does not work in Micropython (i.e., [::1]).

I have since quit the project. The Raspberry Pi Pico doesn't have enough memory for such a big screen, at least not with Micropython.

peterhinch, 2024-12-06:
I think you mean `[::-1]` but you're right in saying it's unsupported in MP. One thought on text reversal is
```py
def reverse(s):
    l = list(s)
    l.reverse()
    return "".join(l)
```
Or
```py
def rev(s):
    if (l := len(s)) == 1:
        return s
    l1 = l // 2
    return ''.join((rev(s[l1:]), rev(s[:l1])))
```
I have avoided using multi-color ePaper displays because of the RAM requirement and extremely slow updates. However the Pico 2 might work, it has 500K of RAM.

regevt, 2024-12-06:
Yeah, I need Wi-Fi, so the Pico 2 is not good enough for my project, and the Pico 2 W is not available yet in the Netherlands.
To be honest, I think the main issue is Waveshare, every display of them that I used had problems that required me to Google for days.

Can you recommend a big (5+) display from a different manufacturer?

peterhinch, 2024-12-06:
[See this list](https://github.com/peterhinch/micropython-nano-gui/blob/master/DISPLAYS.md) which comprises the displays I support. In general Adafruit provide excellent support with well written code (albeit for CircuitPython).

Wveshare's software is not the best. I have ported several of their drivers to MP. In every case making an efficient port took considerable effort. At least they seem to have learned how to make hardware. Their early efforts were not good.

sgbmzm, 2024-12-17:
Thanks!
The Hebrew set is missing one letter (ן)
I am attaching a corrected "Hebrew" file (that also has an additional character or two added).
I am also attaching a file of a ready-made Hebrew font "mriam12_heb.py" but because I was unable to attach a Python file I am only attaching its content.
And also a function for the direction of the Hebrew text, because it is from right to left
[Hebrew.txt](https://github.com/user-attachments/files/18170312/Hebrew.txt)

```
def reverse(string):
    return "".join(reversed(string))

```

```
# Code generated by font_to_py.py.
# Font: mriam.ttf Char set:  !"#$%&'()*+,-./0123456789:ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz{}°אבגדהוזחטיךכלםמןנסעףפץצקרשת
# Cmd: font_to_py.py -k Hebrew.txt mriam.ttf 12 myfont.py
version = '0.42'

def height():
    return 12

def baseline():
    return 10

def max_width():
    return 11

def hmap():
    return True

def reverse():
    return False

def monospaced():
    return False

def min_ch():
    return 32

def max_ch():
    return 1514

_font =\
b'\x05\x00\x00\x00\x60\x90\x10\x10\x20\x20\x00\x20\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x40\x40\x40\x40\x40\x40\x00\x40\x00\x00\x00\x00'\
b'\x05\x00\x00\x00\x50\x50\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x07\x00\x00\x00\x14\x14\x14\xfe\x28\xfe\x50\x50\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x10\x38\x54\x50\x38\x14\x94\x78\x10\x00\x00\x00'\
b'\x09\x00\x00\x00\x00\x00\x02\x00\x34\x00\x4c\x00\x48\x00\x3b\x00'\
b'\x14\x80\x24\x80\x23\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x08\x00\x00\x00\x30\x48\x50\x26\x74\x94\x88\xf7\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x40\x40\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x20\x40\x40\x40\x40\x40\x40\x40\x20\x00\x00\x00'\
b'\x04\x00\x00\x00\x40\x40\x20\x20\x20\x20\x20\x40\x40\x00\x00\x00'\
b'\x06\x00\x10\x54\x38\x7c\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x08\x00\x00\x00\x00\x08\x08\x08\x7f\x08\x08\x08\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x20\x40\x40\x00\x00\x00'\
b'\x04\x00\x00\x00\x00\x00\x00\x00\xe0\x00\x00\x00\x00\x00\x00\x00'\
b'\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x40\x00\x00\x00\x00'\
b'\x05\x00\x00\x10\x20\x20\x20\x40\x40\x40\x80\x80\x80\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x30\x48\x48\x48\x48\x48\x30\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x30\x50\x10\x10\x10\x10\x10\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x30\x48\x08\x10\x20\x40\x78\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x78\x10\x30\x08\x08\x88\x70\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x18\x28\x28\x48\x7c\x08\x08\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x78\x40\xf0\x08\x08\x88\x70\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\x30\x40\x70\x48\x48\x48\x30\x00\x00\x00\x00'\
b'\x06\x00\x00\x00\x00\xf8\x08\x10\x10\x10\x20\x20\x00\x00\x00\x00'\
[... 120 lines cut ...]
b'\xe5\x05\xe0\x00\xe6\x05\xe2\x00\xe7\x05\xe4\x00\xe8\x05\xe6\x00'\
b'\xe9\x05\xe8\x00\xea\x05\xea\x00'

_mvfont = memoryview(_font)
_mvsp = memoryview(_sparse)
ifb = lambda l : l[0] | (l[1] << 8)

def bs(lst, val):
    while True:
        m = (len(lst) & ~ 7) >> 1
        v = ifb(lst[m:])
        if v == val:
            return ifb(lst[m + 2:])
        if not m:
            return 0
        lst = lst[m:] if v < val else lst[:m]

def get_ch(ch):
    doff = bs(_mvsp, ord(ch)) << 3
    width = ifb(_mvfont[doff : ])

    next_offs = doff + 2 + ((width - 1)//8 + 1) * 12
    return _mvfont[doff + 2:next_offs], 12, width

```

peterhinch, 2024-12-18:
Thank you. I'll post `Hebrew.txt` but I'm not keen on publishing submitted pre-built fonts because of potential copyright issues. The idea behind font-to-py is that users build their own (in a size to suit them). That way users take responsibility for copyright :)

closed 2024-12-18 by peterhinch