peterhinch/micropython-font-to-py #68: "python" and "txt"

sgbmzm, 2024-12-17:
Thank you very much for this library!

In the explanation of Font to Python, the example is missing a call to **"Python"** OR **"python3"** so I wasted a few hours until I realized that I need to call Python first.
Also, the extension file name needs to have the extension **".txt"**

`$ font_to_py.py FreeSans.ttf 23 myfont.py`
`$ font_to_py.py -k extended FreeSans.ttf 23 my_extended_font.py`

TO:

`$ python font_to_py.py FreeSans.ttf 23 myfont.py` # OR python3
`$ python font_to_py.py -k extended.txt FreeSans.ttf 23 my_extended_font.py`

It is also worth emphasizing that it is not mandatory to execute both lines in the example, but one of them is possible

peterhinch, 2024-12-17:
If your system is properly configured there should be no need to explicitly call **python**. The first line of code is
```py
#! /usr/bin/env python3
```
the purpose of which is [explained here](https://stackoverflow.com/questions/2429511/why-do-people-write-usr-bin-env-python-on-the-first-line-of-a-python-script). I can run it as follows
```bash
[adminpete@meerkat]: ~
$ font_to_py.py --help
usage: /mnt/qnap2/Scripts/font_to_py.py [-h] [-x] [-y] [-r] [-f] [-b] [-i] [-s SMALLEST] [-l LARGEST] [-e ERRCHAR] [-c CHARSET] [-k CHARSET_FILE]
                                        infile height outfile

font_to_py.py V0.4.0
Utility to convert ttf, otf, bdf and pcf font files to Python source.
Sample usage:
font_to_py.py FreeSans.ttf 23 freesans.py

This creates a font with nominal height 23 pixels with these defaults:
Mapping is horizontal, pitch variable, character set 32-126 inclusive.
Illegal characters will be rendered as "?".

To specify monospaced rendering issue:
font_to_py.py FreeSans.ttf 23 --fixed freesans.py

positional arguments:
  infile                Input file path
  height                Font height in pixels
  outfile               Path and name of output file

options:
  -h, --help            show this help message and exit
  -x, --xmap            Horizontal (x) mapping
  -y, --ymap            Vertical (y) mapping
  -r, --reverse         Bit reversal
  -f, --fixed           Fixed width (monospaced) font
  -b, --binary          Produce binary (random access) font file.
  -i, --iterate         Include generator function to iterate over character set.
  -s SMALLEST, --smallest SMALLEST
                        Ordinal value of smallest character default 32
  -l LARGEST, --largest LARGEST
                        Ordinal value of largest character default 126
  -e ERRCHAR, --errchar ERRCHAR
                        Ordinal value of error character default 63 ("?")
  -c CHARSET, --charset CHARSET
                        Character set. e.g. 1234567890: to restrict for a clock display.
  -k CHARSET_FILE, --charset_file CHARSET_FILE
                        File containing charset e.g. cyrillic_subset.
```

I don't recognise the need for a `.txt` extension. None of the files [here](https://github.com/peterhinch/micropython-font-to-py/tree/master/charsets) have such an extension and all have been tested.

sgbmzm, 2024-12-17:
When I download the "extended" file, it downloads as a "txt" file
And in any case, nothing worked for me without setting the two settings above, that is, adding "python" and also adding .txt

peterhinch, 2024-12-18:
> When I download the "extended" file, it downloads as a "txt" file

How odd. Try using `git clone` to copy the repository. That will preserve filenames.

Re `python` The only other thing I can think of is setting execute permission on `font_to_py.py`. Without access to your PC it's hard for me to diagnose this.

sgbmzm, 2024-12-19:
I think the reason is that I need to use python instead of python3, so if it had been written `#! /usr/bin/env python` it would have worked.

peterhinch, 2024-12-19:
Interesting. On some machines `#! /usr/bin/env python` invoked python2 which doesn't work.