peterhinch/micropython-font-to-py #70: Bitmap editor

sk8board, 2024-12-30:
Thank you for sharing this excellent work.

I noticed [there](https://github.com/peterhinch/micropython-font-to-py/blob/master/icon_fonts/README.md#the-bitmap-editor) is mention of a bitmap editor which uses the following command:

[image]

I cannot find the file to accept this command. Could you help with how to generate these `bitmap.c` files?

sk8board, 2024-12-30:
FYI: I found the bitmap software documentation [here](https://www.x.org/releases/X11R6.7.0/doc/bitmap.1.html).

closed 2024-12-30 by sk8board

peterhinch, 2024-12-30:
Perhaps I assumed too much Unix knowledge when I wrote
> This is documented in the man pages

If you enter
```bash
$ man bitmap
```
you will get docs.

sk8board, 2024-12-30:
Yes, but I now understand. Thank you!

* I have created custom icons 20x20 with the linux bitmap software. 
* I save the file as icon.c, 
* then I update the filnames.txt. 
* Next I use the `c_to_python_font.py` to create a `font.py` file. 

The result does not look like the icon which I created. If I remake the `font.py` file, the result is different, but still does not look like the icon which I created. 

I am using [a driver](https://github.com/burgrp/micropython-ili9225), which references this [micropython-font-to-py](https://github.com/peterhinch/micropython-font-to-py) GitHub. Any ideas why my icon result does not look like what I created?

peterhinch, 2024-12-31:
It's hard for me to comment on display drivers which are not in the [supported drivers list](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md), not least because I don't own suitable hardware to test.

The design of the driver you are using is very different from the supported drivers. It also has its own way of supporting Python fonts unrelated to the `Writer` class. I can only suggest you take this up with the driver's author.

peterhinch, 2024-12-31:
To add to this, I converted your C file to a font and displayed it on a supported display and it looked fine.

sk8board, 2025-01-05:
I found the reason for the icon image was being displayed incorrect is because the driver needed the bit order reversed for each byte.

I noticed `font_to_py.py` accepts [an argument](https://github.com/peterhinch/micropython-font-to-py/blob/master/FONT_TO_PY.md#312-optional-arguments) `-r` to reverse the order of each font byte. Does `c_to_python_font.py` accept a similar argument?

peterhinch, 2025-01-05:
The short answer is no.

Development of the `font_to_py.py` utility started over eight years ago, before `FrameBuffer`. Its output was intended to be compatible with a range of displays, some of which had odd mappings.

When `FrameBuffer` was developed I started work on the three GUI's. Device drivers are subclassed from `FrameBuffer`. Fonts are specified with horizontal mapping so that a font can be used with any GUI and any supported display. Any bit-twiddling is done in the display driver.

Consequently all my work on image display and icons is based on horizontal mapping. The other mappings in `font_to_py.py` exist for historical reasons and remain for experimenters.