peterhinch/micropython-font-to-py #72: Writer in Pico_ePaper-2.9-B.py

brylka787, 2025-02-14:
Hello Peter,

could you improve the

https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/python/Pico_ePaper-2.9-B.py

with Writer.

I need larger fonts.

And the language Serbian.

Created with font_to_py.py.

In the .PY file I only need EPD_2in9_Landscape

without display_Partial

without v/hline, line, rect

And without image4Gray

Regards
Peter

peterhinch, 2025-02-15:
That Waveshare driver is unsuitable for `Writer` because the driver is not subclassed from `FrameBuffer`. Unfortunately I cannot write a new driver for a display which I don't possess.

If you're prepared to use one of the many [supported displays](https://github.com/peterhinch/micropython-nano-gui/blob/master/DISPLAYS.md) you will easily be able to create and use Serbian fonts.

closed 2025-02-15 by peterhinch

brylka787, 2025-02-15:
Hello Peter,

This display works with Writer.

In:
https://github.com/waveshareteam/Pico_ePaper_Code/blob/main/pythonNanoGui/demos/ePaper_test.py

In line 16 I can enter fonts.Serbian24. Serbian24 is created with font_to_py.py and
copied into the gui.fonts folder.

In line 40 I replace 'short' with my 'Serbian text' and
cyrilic is displayed with a height of 24 pixels.

What I don't understand is why it says 'longer' there.?

And where I can add more lines.

In the color_setup.py I uncommented line 35 and
commented 25 and 26.

But this pythonNanoGui from the waveshareteam is too complicated and too time-consuming for me.

Peter

brylka787, 2025-02-15:
and
In color_setup.py I changed lines 61 to 63.

ssd = SSD(spi, pcs, pdc, prst, pbusy, landscape=False, asyn=False)  # Create a display instance
#ssdred = SSDred(spi, pcs, pdc, prst, pbusy, landscape=False)  # Cread a red display instance (just for B model)
ssd.demo_mode = True

peterhinch, 2025-02-16:
The drivers in Waveshare's `PythonNanoGui`  tree are written to be compatible with nano-gui. Consequently, these drivers will also be compatible with the `Writer` classes. You are free to use `Writer` on its own or nano-gui. Whichever you use I suggest you read my documentation - docs are not Waveshare's strong point.